ros-controls/realtime_tools #23: Hard coded loop rate

jlack1987, 2017-11-16:
Forgive my ignorance if my issue is interpreting the code incorrectly as i'm pretty new to realtime_tools. In [this line](https://github.com/ros-controls/realtime_tools/blob/a5467ed2371abaab7fa646a51a9db72bc8b4771f/src/realtime_clock.cpp#L103) the loop rate is hard coded to 750Hz. To me this means that if you're running at loop rates greater than 750Hz then you should probably not use this clock tool correct? It seems it would be very easy to make this a configurable argument perhaps with a default rate of 750Hz to not break backwards compatibility.

bmagyar, 2017-11-17:
Seems like a pretty good point. I personally haven't hit that limit yet. 

@graiola , haven't you guys ran stuff on Meka at 1KHz? Perhaps you didn't use this class?

ipa-mdl, 2017-11-17:
I have not found any user on GitHub for this class, only file copies.
I am not sure if `ros::Time::now()` is using a syscall underneath.

However, I don't see any reason why this should be hard-coded.

jlack1987, 2017-11-17:
Just wondering bc I just stumbled upon this class and was a bit surprised to see, what looks to me, a wrapper around ```ros::Time``` to make it real time safe when I wasn't aware that it ever was not real time safe. It got me thinking that for my own stuff perhaps I should be swapping out my calls using regular ole ```ros::Time``` to using this for better RT performance.

ipa-mdl, 2017-11-17:
it uses `gettimeofday`, which will make as syscall and so might lead to context switch.
I guess this is the reason for passing the time to the update function in `ros_control`.

jlack1987, 2017-11-17:
Ah ok that was what I didn't know, thanks for the explanation. Based on the source code chunk here,
```
#if HAS_CLOCK_GETTIME
           timespec start;
           clock_gettime(CLOCK_REALTIME, &start);
           sec  = start.tv_sec;
           nsec = start.tv_nsec;
         #else
           struct timeval timeofday;
           gettimeofday(&timeofday,NULL);
           sec  = timeofday.tv_sec;
           nsec = timeofday.tv_usec * 1000;
         #endif
```
it looks like we are probably ok as long as ```HAS_CLOCK_GETTIME``` is defined.

graiola, 2017-11-17:
If I remember correctly `ros::Time::now();` is not real-time safe.
You should use the time implementation provided by your rtos and pass the time information to the update function in `ros_control`. For example (with RTAI):
```
const uint64_t one_E9 = 1000000000ULL;
uint64_t nsec64 = rt_get_cpu_time_ns();
uint32_t sec32_part = nsec64 / one_E9;
uint32_t nsec32_part = nsec64 % one_E9;
ros::Time now(sec32_part, nsec32_part);
cm_ptr_->update(now, period_);
```

graiola, 2017-11-17:
I closed it by mistate, sorry.

jlack1987, 2017-11-17:
Thanks so much this is really good info, and sorry for sidetracking things from the issue's real purpose. If you all are interested I can make a PR to add that loop rate as an arg with default value of 750.

bmagyar, 2017-11-17:
This is something that could potentially be in packages typically not
released to the public. This could explain why it is hard to find usage of
this. Still... I think adding that param can't hurt.

bmagyar, 2020-12-05:
Happy to review a PR to add that loop rate as an arg with default value of 750.

PR #261 (2025-01-30): Remove RealtimeClock
RealtimeClock got deprecated on Jazzy and Rolling with their sync within the last days.

Fixes #241 and closes #23
closed 2025-01-30 by christophfroehlich