ros-controls/realtime_tools #35: Porting to ROS 2 Dashing

sloretz, 2019-05-22:
I'm looking into how much work it would be to port this package into ROS 2 Dashing. Mind answering a few questions?

* Is someone already working on this?
* Where should a ros2 port live? How about a `dashing-devel` branch in this repo?
* Would the maintainers be willing to release into Dashing once code has been ported?
* Should the port also eliminate the use of boost? https://github.com/ros-controls/realtime_tools/pull/31#issuecomment-452314598
* Is a partial port OK?
  * For example, I would like to use `RealtimePublisher`, but I don't currently need `RealtimeServerGoalHandle`. Would porting these classes one at a time be acceptable?
* Should the existing code style be maintained, or should the [ROS 2 style guide](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#id1) be used?

bmagyar, 2019-05-23:
Hi @sloretz , 
Great list, allow me to answer in the same fashion.

* Not that I know of.
* Works perfectly fine.
* No problem.
* As much as possible, I've been trying to get rid of most boost in favour of standard library.
* I think a partial port may be misleading and harder to maintain. This repo doesn't have that much code TBH. It may be a bigger issue elsewhere.
* Feel free to restyle things, but please try doing it in a separate PR from functional ones. I have only one problem with the style guide: I find `CamelCase` for functions terrible, also most of our existing API uses `camelCase` which is just way more readable and avoids confusing functions with classes.

Hope this helps!

ipa-mdl, 2019-05-23:
> I find CamelCase for functions terrible, also most of our existing API uses camelCase which is just way more readable and avoids confusing functions with classes.

Urgs, I did not see this before. :+1:  for `camelCase`.

sloretz, 2019-09-04:
@bmagyar now that #41 is merged :tada:, would you be willing to release realtime_tools into dashing?

bmagyar, 2019-09-09:
Sorry for the slight delay, here you go: https://github.com/ros/rosdistro/pull/22203

PR #41 (2019-08-28): Port to Dashing
Resolves #35. The tests all pass. It has been rebased on 6e0e91ef34eb16b3aa95987d35ef65ee7060ea2b, which `dashing-devel` could be fast-forwarded to.

Misc changes:
* `RealtimeClock` and `RealtimeServerGoalHandle` now take an `rclcpp::Logger`
  * If no logger is given then they use one named `"realtime_tools"`
* `RealtimeServerGoalHandle` has a new method `execute()`
* `RealtimeClock` now takes an `rclcpp::Clock`
* Added `RealtimeClock::now()` and deprecated `RealtimeClock::getSystemTime()` because the clock may not be a system time clock and `now()` matches `rclcpp::Clock::now( [... cut]
closed 2019-11-20 by sloretz