ros-controls/realtime_tools #44: unable to build on arch linux due to gmock not being found

jwhendy, 2019-09-09:
I posted [an issue](https://github.com/ros/catkin/issues/1030) at ros/catkin as well, since I first found issues related to gtest/gmock there. Downloading current from git and building via `catkin build` yields the following output for me:
```
Errors     << realtime_tools:cmake /home/jwhendy/catkin_ws/logs/realtime_tools/build.cmake.000.log                                                                                       
CMake Warning at /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:151 (message):
  skipping gmock 'realtime_box_tests' in project 'realtime_tools' because
  gmock was not found
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:80 (_catkin_add_executable_with_google_test)
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:54 (_catkin_add_google_test)
  CMakeLists.txt:23 (catkin_add_gmock)

CMake Error at /home/jwhendy/catkin_ws/src/realtime_tools/CMakeLists.txt:24 (target_link_libraries):
  Cannot specify link libraries for target "realtime_box_tests" which is not
  built by this project.

cd /home/jwhendy/catkin_ws/build/realtime_tools; catkin build --get-env realtime_tools | catkin env -si  /usr/bin/cmake /home/jwhendy/catkin_ws/src/realtime_tools --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/jwhendy/catkin_ws/devel/.private/realtime_tools -DCMAKE_INSTALL_PREFIX=/home/jwhendy/catkin_ws/install -DPYTHON_EXECUTABLE=/usr/bin/python2 -DPYTHON_INCLUDE_DIR=/usr/include/python2.7 -DPYTHON_LIBRARY=/usr/lib/libpython2.7.so; cd -
```

Also, I do have them installed:
```
$ pacman -Qi gmock
Name            : gmock
Version         : 1.8.1-3

$ pacman -Qi gtest
Name            : gtest
Version         : 1.8.1-3

$ ls /usr/include/gmock/
internal         gmock-cardinalities.h      gmock-generated-function-mockers.h  gmock-generated-nice-strict.h  gmock-matchers.h      gmock-more-matchers.h
gmock-actions.h  gmock-generated-actions.h  gmock-generated-matchers.h          gmock.h                        gmock-more-actions.h  gmock-spec-builders.h
```

Any other details which could help troubleshoot this?

jwhendy, 2019-09-14:
My build attempt is part of an overall effort to get ROS packages building on Arch linux, and we're migrating to upstream sources vs. the former practice of using [ros-gbp](https://github.com/ros-gbp). I reverted to the ros-gbp PKGBUILD, and found that [1.13.1](https://github.com/ros-gbp/realtime_tools-release/commit/8ff575052bd4b4d2f856d6d2b37bb6addf1ff8ab) still builds. 

Not sure if it's helpful, but when I look at the src files for each, I find:
```
### 1.13.1 from ros-gbp: successful

$ grep -ri gmock *
src/build/CMakeCache.txt://The directory containing a CMake configuration file for GMock.
src/build/CMakeCache.txt:GMock_DIR:PATH=GMock_DIR-NOTFOUND
src/build/CMakeCache.txt:_gmock_INCLUDES:FILEPATH=_gmock_INCLUDES-NOTFOUND
src/build/CMakeCache.txt:_gmock_SOURCES:FILEPATH=_gmock_SOURCES-NOTFOUND

$ grep -ri gtest *
src/build/CMakeFiles/Makefile.cmake:  "/opt/ros/melodic/share/catkin/cmake/test/gtest.cmake"
src/build/CMakeFiles/Makefile.cmake:  "/usr/share/cmake-3.15/Modules/FindGTest.cmake"
src/build/catkin/catkin_generated/version/package.cmake:set(catkin_BUILD_EXPORT_DEPENDS "google-mock" "gtest" "python-empy" "python-nose" "python-argparse" "python-catkin-pkg")
src/build/catkin/catkin_generated/version/package.cmake:set(catkin_RUN_DEPENDS "python-argparse" "python-catkin-pkg" "google-mock" "gtest" "python-empy" "python-nose")
src/build/CMakeCache.txt:GTEST_INCLUDE_DIR:PATH=/usr/include
src/build/CMakeCache.txt:GTEST_LIBRARY:FILEPATH=/usr/lib/libgtest.so
src/build/CMakeCache.txt:GTEST_LIBRARY_DEBUG:FILEPATH=GTEST_LIBRARY_DEBUG-NOTFOUND
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARY:FILEPATH=/usr/lib/libgtest_main.so
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARY_DEBUG:FILEPATH=GTEST_MAIN_LIBRARY_DEBUG-NOTFOUND
src/build/CMakeCache.txt:_gtest_INCLUDES:FILEPATH=_gtest_INCLUDES-NOTFOUND
src/build/CMakeCache.txt:_gtest_SOURCES:FILEPATH=/usr/src/gtest/src/gtest.cc
src/build/CMakeCache.txt:GTEST_BOTH_LIBRARIES:INTERNAL=/usr/lib/libgtest.so;/usr/lib/libgtest_main.so
src/build/CMakeCache.txt:GTEST_FOUND:INTERNAL=TRUE
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_INCLUDE_DIR
src/build/CMakeCache.txt:GTEST_INCLUDE_DIR-ADVANCED:INTERNAL=1
src/build/CMakeCache.txt:GTEST_INCLUDE_DIRS:INTERNAL=/usr/include
src/build/CMakeCache.txt:GTEST_LIBRARIES:INTERNAL=/usr/lib/libgtest.so
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_LIBRARY
src/build/CMakeCache.txt:GTEST_LIBRARY-ADVANCED:INTERNAL=1
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_LIBRARY_DEBUG
src/build/CMakeCache.txt:GTEST_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARIES:INTERNAL=/usr/lib/libgtest_main.so
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_MAIN_LIBRARY
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARY-ADVANCED:INTERNAL=1
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_MAIN_LIBRARY_DEBUG
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARY_DEBUG-ADVANCED:INTERNAL=1

### building from  from ros-gbp: successful
$ grep -ri gmock *
src/realtime_tools-1.15.0/CHANGELOG.rst:* Switch to gmock
src/realtime_tools-1.15.0/CMakeLists.txt:  catkin_add_gmock(realtime_box_tests test/realtime_box_tests.cpp)
src/realtime_tools-1.15.0/CMakeLists.txt:  target_link_libraries(realtime_box_tests ${PROJECT_NAME} ${GMOCK_MAIN_LIBRARIES})
src/realtime_tools-1.15.0/CMakeLists.txt:  catkin_add_gmock(realtime_buffer_tests test/realtime_buffer_tests.cpp)
src/realtime_tools-1.15.0/CMakeLists.txt:  target_link_libraries(realtime_buffer_tests ${PROJECT_NAME} ${GMOCK_MAIN_LIBRARIES})
src/realtime_tools-1.15.0/CMakeLists.txt:  catkin_add_gmock(realtime_clock_tests test/realtime_clock_tests.cpp)
src/realtime_tools-1.15.0/CMakeLists.txt:  add_rostest_gmock(realtime_publisher_tests test/realtime_publisher.test test/realtime_publisher_tests.cpp)
src/realtime_tools-1.15.0/CMakeLists.txt:  add_rostest_gmock(realtime_server_goal_handle_tests test/realtime_server_goal_handle.test test/realtime_server_goal_handle_tests.cpp)
src/realtime_tools-1.15.0/test/realtime_box_tests.cpp:#include <gmock/gmock.h>
src/realtime_tools-1.15.0/test/realtime_buffer_tests.cpp:#include <gmock/gmock.h>
src/realtime_tools-1.15.0/test/realtime_clock_tests.cpp:#include <gmock/gmock.h>
src/realtime_tools-1.15.0/test/realtime_publisher_tests.cpp:#include <gmock/gmock.h>
src/realtime_tools-1.15.0/test/realtime_server_goal_handle_tests.cpp:#include <gmock/gmock.h>
src/build/CMakeCache.txt://The directory containing a CMake configuration file for GMock.
src/build/CMakeCache.txt:GMock_DIR:PATH=GMock_DIR-NOTFOUND
src/build/CMakeCache.txt:_gmock_INCLUDES:FILEPATH=_gmock_INCLUDES-NOTFOUND
src/build/CMakeCache.txt:_gmock_SOURCES:FILEPATH=_gmock_SOURCES-NOTFOUND

$ grep -ri gtest *
src/build/CMakeFiles/Makefile.cmake:  "/opt/ros/melodic/share/catkin/cmake/test/gtest.cmake"
src/build/CMakeFiles/Makefile.cmake:  "/usr/share/cmake-3.15/Modules/FindGTest.cmake"
src/build/catkin/catkin_generated/version/package.cmake:set(catkin_BUILD_EXPORT_DEPENDS "google-mock" "gtest" "python-empy" "python-nose" "python-argparse" "python-catkin-pkg")
src/build/catkin/catkin_generated/version/package.cmake:set(catkin_RUN_DEPENDS "python-argparse" "python-catkin-pkg" "google-mock" "gtest" "python-empy" "python-nose")
src/build/CMakeCache.txt:GTEST_INCLUDE_DIR:PATH=/usr/include
src/build/CMakeCache.txt:GTEST_LIBRARY:FILEPATH=/usr/lib/libgtest.so
src/build/CMakeCache.txt:GTEST_LIBRARY_DEBUG:FILEPATH=GTEST_LIBRARY_DEBUG-NOTFOUND
[... cut ...]
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_LIBRARY_DEBUG
src/build/CMakeCache.txt:GTEST_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARIES:INTERNAL=/usr/lib/libgtest_main.so
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_MAIN_LIBRARY
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARY-ADVANCED:INTERNAL=1
src/build/CMakeCache.txt://ADVANCED property for variable: GTEST_MAIN_LIBRARY_DEBUG
src/build/CMakeCache.txt:GTEST_MAIN_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
```

bmagyar, 2019-09-16:
Is this related to only `realtime_tools`? Can you build any other package what has testing?

jwhendy, 2019-09-16:
If you suggest a package that also needs `gmock` testing, I'll try. This is the only package I've run into this with.

Not sure if this is helpful, but [mavros](https://github.com/mavlink/mavros/issues/1280) had a realated problem as well. Maybe looking at commonalities would be helpful?

As mentioned above, I was able to build the ros-gbp archive version, so I think it's been a change since then, either here or with `catkin`.

ipa-mdl, 2019-09-16:
`catkin` has `build_export_depend` on `google-mock`, so it should be available in all catkin-based environments. I can't see any recent changes..

jwhendy, 2019-09-17:
> gmock tests in realtime_toolswere introduced some time ago.

Indeed, at least 1.15.0 if not earlier. This is what I find as well (mentioned above), namely that 1.13.1 builds but not 1.15.0.

```
[realtime_tools-1.13.1]$ grep -r mock *
# nothing

[realtime_tools-1.15.0]$ grep -r mock *
CHANGELOG.rst:* Switch to gmock
CMakeLists.txt:  catkin_add_gmock(realtime_box_tests test/realtime_box_tests.cpp)
CMakeLists.txt:  catkin_add_gmock(realtime_buffer_tests test/realtime_buffer_tests.cpp)
CMakeLists.txt:  catkin_add_gmock(realtime_clock_tests test/realtime_clock_tests.cpp)
CMakeLists.txt:  add_rostest_gmock(realtime_publisher_tests test/realtime_publisher.test test/realtime_publisher_tests.cpp)
CMakeLists.txt:  add_rostest_gmock(realtime_server_goal_handle_tests test/realtime_server_goal_handle.test test/realtime_server_goal_handle_tests.cpp
```

I think this is likely about how `catkin` is looking for `gmock` given [#1022](https://github.com/ros/catkin/pull/1022) or how [arch installs gmock/gtest](https://bugs.archlinux.org/task/63793?project=5&string=gmock) (based on some [helpful info](https://github.com/ros/catkin/issues/1030#issuecomment-531868346) of the system components on Debian).

With the posts on the [issue at catkin](https://github.com/ros/catkin/issues/1030), I think this is likely about catkin or my system. You can probably safely close this if you like, otherwise leave it open and I'll post back with the conclusion.

This is still the only package where I've run into this, and I've probably built ~100+ packages from arch's user repository over the past week. Either others are fine, or hardly anyone is using `gmock`...

bionade24, 2020-01-15:
It would be nice if somethig happens or we still can get some information about this.

acxz, 2020-02-16:
Just tried to build the latest release 1.15.1, but I am still getting the same error.

bionade24, 2020-02-16:
Ok, has anyone ever tried to get help on  answers.ros.org ? Maybe someone has a patch for it. Does realtime_tools work if installed via `wstool` and `catkin_make` or `catkin build` ?

matthew-reynolds, 2020-02-16:
I believe this is an upstream issue with catkin, being tracked [here](https://github.com/ros/catkin/issues/1030).

jwhendy, 2020-02-17:
@matthew-reynolds is correct. My assessment from the issue he linked (which is also mine) is that this is about `gmock`, not `realtime_tools`. I forgot I opened this when I ran into that, and will now close.

closed 2020-02-17 by jwhendy