ros-controls/realtime_tools #45: Errors with realtime_tools during catkin_make

Pranav24-8, 2019-10-22:
Hi,
I am getting the following error during catkin_make. Please help. It looks like the error is because of the latest commit in the repository file realtime_server_goal_handle.h.
```bash
In file included from /root/catkin_ws/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:66:0,
                 from /root/catkin_ws/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:60:3: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   ResultPtr preallocated_result_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~
   Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:61:3: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   FeedbackPtr preallocated_feedback_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~~~
   Feedback
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:50: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                  ^~~~~~~~~
                                                  Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:115: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                                                                                   ^~~~~~~~~~~
                                                                                                                   Feedback
In file included from /root/catkin_ws/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:66:0,
                 from /root/catkin_ws/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h: In constructor ‘realtime_tools::RealtimeServerGoalHandle<Action>::RealtimeServerGoalHandle(realtime_tools::RealtimeServerGoalHandle<Action>::GoalHandle&, const int&, const int&)’:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:68:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_result_’
       preallocated_result_(preallocated_result),
       ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:69:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_feedback_’
       preallocated_feedback_(preallocated_feedback)
       ^~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:71:10: error: ‘preallocated_result_’ was not declared in this scope
     if (!preallocated_result_)
          ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:71:10: note: suggested alternative: ‘preallocated_result’
     if (!preallocated_result_)
          ^~~~~~~~~~~~~~~~~~~~
          preallocated_result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:73:10: error: ‘preallocated_feedback_’ was not declared in this scope
     if (!preallocated_feedback_)
          ^~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:73:10: note: suggested alternative: ‘preallocated_feedback’
     if (!preallocated_feedback_)
          ^~~~~~~~~~~~~~~~~~~~~~
          preallocated_feedback
In file included from /root/catkin_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:56:0,
                 from /root/catkin_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:60:3: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   ResultPtr preallocated_result_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~
   Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:61:3: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   FeedbackPtr preallocated_feedback_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~~~
   Feedback
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:50: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                  ^~~~~~~~~
                                                  Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:115: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                                                                                   ^~~~~~~~~~~
                                                                                                                   Feedback
In file included from /root/catkin_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:56:0,
                 from /root/catkin_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h: In constructor ‘realtime_tools::RealtimeServerGoalHandle<Action>::RealtimeServerGoalHandle(realtime_tools::RealtimeServerGoalHandle<Action>::GoalHandle&, const int&, const int&)’:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:68:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_result_’
       preallocated_result_(preallocated_result),
       ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:69:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_feedback_’
       preallocated_feedback_(preallocated_feedback)
[... 299 lines cut ...]
/root/catkin_ws/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h:630:33: note:   when instantiating default argument for call to realtime_tools::RealtimeServerGoalHandle<Action>::RealtimeServerGoalHandle(realtime_tools::RealtimeServerGoalHandle<Action>::GoalHandle&, const int&, const int&) [with Action = control_msgs::FollowJointTrajectoryAction_<std::allocator<void> >; realtime_tools::RealtimeServerGoalHandle<Action>::GoalHandle = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >]
   RealtimeGoalHandlePtr rt_goal(new RealtimeGoalHandle(gh));
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/catkin_ws/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:66:0,
                 from /root/catkin_ws/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:163: error: ‘FeedbackPtr’ was not declared in this scope
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                                                                                                                        ~~~~~~~~~~~^~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:163: note: suggested alternative: ‘Feedback’
[... cut ...]
make[2]: *** [ros_controllers/gripper_action_controller/CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o] Error 1
ros_controllers/gripper_action_controller/CMakeFiles/gripper_action_controller.dir/build.make:62: recipe for target 'ros_controllers/gripper_action_controller/CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o' failed
make[1]: *** [ros_controllers/gripper_action_controller/CMakeFiles/gripper_action_controller.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

bmagyar, 2019-10-22:
Please switch `realtime_tools` to the `melodic-devel` branch.

Pranav24-8, 2019-10-22:
I'm using apt install to install this. How do I switch branch then??

bmagyar, 2019-10-22:
There you don't :) I'm happy to say though that you are safe, in this case you have `ros_controllers` checked out from `kinetic-devel`, not `melodic-devel`. This is expected, just make sure you use the right branch for the ROS distro you are running. You can check on the main page of `ros_controllers` that all build branches are green, the issues is only with your checkout.

Pranav24-8, 2019-10-22:
Right I'll check it out. Thank You for the help.

matthew-reynolds, 2020-02-16:
@Pranav24-8 is this resolved? If so, let's close this issue.

Pranav24-8, 2020-02-16:
Thank You. The issue is resolved.

closed 2020-02-16 by Pranav24-8