ros-controls/realtime_tools #50: volatile bool -> atomic<bool>

Maverobot, 2020-02-09:
When reading the code, I realized in [realtime_publisher.h](https://github.com/ros-controls/realtime_tools/blob/melodic-devel/include/realtime_tools/realtime_publisher.h), `volatile bool` is used for cross-threads flags. As far as I know, it is undefined behavior in C++. Should we replace that with `std::atomic<bool>`? 

Some discussions on Stackoverflow:
* [C++ : std::atomic<bool> and volatile bool](https://stackoverflow.com/q/29633222/6018272)
* [Is volatile bool for thread control considered wrong?](https://stackoverflow.com/questions/6995310/is-volatile-bool-for-thread-control-considered-wrong)

matthew-reynolds, 2020-02-16:
1. Indeed, `volatile bool` here is incorrect and dangerous. Looks like something written nearly 10 years ago that's been carried forward ever since. I don't think the solution is `atomic<bool>` though, I think `atomic_flag` is more suited since it is guaranteed to be implemented lock-free, while the `atomic<T>` types may be implemented with locks. (Practically, I imagine all implementation for an `atomic<bool>` are lock-free, but we should still strive for correctness.)
Looks like #51 was just opened to address this issue - Let's move this discussion there?

2. Agreed, `int turn_` also looks like a good candidate for an `std::atomic<>` type, this time `std::atomic<int>`.

3. I don't know why `NON_POLLING` isn't default (@bmagyar thoughts? My only guess is that when it was written in 2013, it required boost and was therefore "heavier"?), but from #43 it looks like it currently doesn't build anyways. Perhaps once it's fixed we should investigate enabling it by default, at least in Noetic.

bmagyar, 2020-03-11:
Thank you guys for chasing this.

I propose the following: 
* #43 to fix the state of affairs in Melodic
* #52 to bring a proper fix and to use the non-polling in Noetic
* #51 to spice up the ROS2 world, here I'd prefer to not disturb Eloquent but rather target Foxy. First and foremost, I'll update the master branch to host the latest ros2 branch and retarget #51 to that, then cut a foxy release

christophfroehlich, 2024-11-23:
Moving the last open point to #199

closed 2024-11-23 by christophfroehlich