ros-controls/realtime_tools #70: New noetic release including #58

timonegk, 2021-01-14:
The current release for noetic does not build because #58 is not included.

bmagyar, 2021-01-14:
Every time you report a build failure, please do the courtesy of reporting the setup you encountered this issue in.

The ROS buildfarm was happy with the release the last time I pushed it. I can however push a new release to Noetic if that's what you are trying to ask for in an indirect way? ;)

timonegk, 2021-01-14:
I'm sorry, I thought it might have been a known problem, hence #58. I am building on Debian Testing where the patch seems to be required.
And yes, I am asking for a new release to noetic :see_no_evil:

bmagyar, 2021-01-14:
https://github.com/ros/rosdistro/pull/27994

bmagyar, 2021-01-14:
have a great day ;)

timonegk, 2021-01-15:
Thanks a lot, you too :) Everything builds now.

closed 2021-01-15 by timonegk

bmagyar, 2021-01-16:
no worries, thanks for reporting!