ros-controls/realtime_tools #96: Why does .rosdistro for Rolling point to the `foxy-devel` branch?

AndyZe, 2023-01-04:
I think this is causing a MoveIt package which depends on a new Rolling feature to fail.

https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml

```
  realtime_tools:
    doc:
      type: git
      url: https://github.com/ros-controls/realtime_tools.git
      version: foxy-devel
    release:
      tags:
        release: release/rolling/{package}/{version}
      url: https://github.com/ros2-gbp/realtime_tools-release.git
      version: 2.4.0-1
```

bmagyar, 2023-01-05:
The responsible person has been apprehended and was forced to fix his mistake: 
https://github.com/ros/rosdistro/pull/35761

AndyZe, 2023-01-05:
🙏, responsible person

closed 2023-01-05 by AndyZe