ros-controls/realtime_tools #131: Build fails on Windows: `Cannot open include file: 'sched.h'`

RobertoRoos, 2023-11-08:
I am trying to build and run the ROS2Control demo, but during building I run into the following error: `Cannot open include file: 'sched.h'`

I am following the instructions here: https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source 
Using ROS2 Humble, with Windows 10.

The full build output is:

<details>
<summary>stdout_stderr.log</summary>
```
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- Found ament_cmake: 1.3.5 (C:/dev/ros2_humble/share/ament_cmake/cmake)
-- Found rclcpp: 16.0.5 (C:/dev/ros2_humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (C:/dev/ros2_humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (C:/dev/ros2_humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (C:/dev/ros2_humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (C:/dev/ros2_humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.3 (C:/dev/ros2_humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rclcpp_action: 16.0.5 (C:/dev/ros2_humble/share/rclcpp_action/cmake)
-- Found ament_cmake_gmock: 1.3.5 (C:/dev/ros2_humble/share/ament_cmake_gmock/cmake)
-- Found test_msgs: 1.2.1 (C:/dev/ros2_humble/share/test_msgs/cmake)
-- Found gmock sources under 'C:/dev/ros2_humble/src/gmock_vendor': C++ tests using 'Google Mock' will be built
-- Found ament_cmake_gtest: 1.3.5 (C:/dev/ros2_humble/share/ament_cmake_gtest/cmake)
-- Found gtest sources under 'C:/dev/ros2_humble/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Configuring done (4.1s)
-- Generating done (1.6s)
-- Build files have been written to: C:/Users/Name/project/ros2_control_ws/build/realtime_tools
MSBuild version 17.7.2+d6990bcfa for .NET Framework

  1>Checking Build System
  gmock.vcxproj -> C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\gmock\Release\gmock.lib
  gmock_main.vcxproj -> C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\gmock\Release\gmock_main.lib
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
  realtime_tools.vcxproj -> C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\Release\realtime_tools.dll
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'Release\realtime_tools.lib' [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\realtime_box_tests.vcxproj]
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'Release\realtime_tools.lib' [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\realtime_buffer_tests.vcxproj]
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'Release\realtime_tools.lib' [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\realtime_clock_tests.vcxproj]
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'Release\realtime_tools.lib' [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\realtime_publisher_tests.vcxproj]
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'Release\realtime_tools.lib' [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\realtime_publisher_tests_non_polling.vcxproj]
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'Release\realtime_tools.lib' [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\realtime_server_goal_handle_tests.vcxproj]
  Building Custom Rule C:/Users/Name/project/ros2_control_ws/src/realtime_tools/CMakeLists.txt
  thread_priority.cpp
C:\Users\Users\Name\project\ros2_control_ws\src\realtime_tools\src\thread_priority.cpp(31,10): fatal  error C1083: Cannot open include file: 'sched.h': No such file or directory [C:\Users\Users\Name\project\ros2_control_ws\build\realtime_tools\thread_priority.vcxproj]

```
</details>

I am not sure to what extend Windows is even supported for ROS2Control, I hope this issue won't do any harm. Thanks!

bmagyar, 2023-11-08:
Windows support: we do have all the boilerplate for DLLs in place. Everything else should be handled by ament but we don't have a windows CI so no means really to keep things building. You are welcome to propose a fix, the Linux CIs should help guide you to a compatible solution.

bmagyar, 2023-11-08:
Please no `#IF Windows` :D

bmagyar, 2023-11-08:
I found a simple solution, using cygwin/mingw (aka anything coming from gcc and the POSIX world) will have compatible versions: https://stackoverflow.com/questions/27667552/porting-sched-h-to-windows

SENAI-GilmarCorreia, 2024-11-13:
Hello @RobertoRoos! I think we solve this problem with the #170 , #179 and #180 PRs. Can you please check if this work to you?

Thank you

RobertoRoos, 2024-11-14:
Great to hear fixes have been merged, but unfortunately I cannot test this anymore, I don't have this software stack anymore.

I'll leave it up to you close this issue.

christophfroehlich, 2024-12-22:
Closing this as we have a windows CI build job to fix issues like these
https://github.com/ros-controls/realtime_tools/actions/workflows/rolling-binary-build-win.yml

closed 2024-12-22 by christophfroehlich