ros-controls/realtime_tools #176: Allow to rename thread in `RealtimePublisher`

roncapat, 2024-10-29:
At least in Linux, it is possible to rename a thread, via calls like [ptrhead_setname_np](https://man7.org/linux/man-pages/man3/pthread_setname_np.3.html).

For application featuring many `RealtimePublishers` and many other threads, diving into `perf record` outputs (e.g. via `hotspot` tool) is difficult because by default all threads have the same name (First 15 characters of the process name), and thread renaming is a powerful tool to discriminate at a glance which thread does what.

To allow this, either the implementation of `RealtimePublisher` shall permit to obtain the native handle (see [this](https://en.cppreference.com/w/cpp/thread/thread/native_handle)) or may enable for Linux only an additional method that uses `pthread_setname_np`.

I can submit a PR depending on maintainers feedback of course 👍🏻

christophfroehlich, 2024-11-23:
@saikishor what do you think?

roncapat, 2024-11-28:
Fine for me, as long as I have a way to access the thread handle :)

firesurfer, 2024-11-28:
@roncapat I suggest you create a PR for that ;)

PR #228 (2024-12-16): Allow to get thread handle in `RealtimePublisher`
Trying to close #176 as suggested by @firesurfer.
closed 2024-12-16 by saikishor