ros-controls/realtime_tools #212: Realtime Publisher - Select one behavior (Polling or NonPolling) and remove the other one

firesurfer, 2024-11-28:
I just refactored the RealtimePublisher in #211 and I am wondering if we really need these two different behaviors of the RT publisher. 

They make the code harder to read and maintain. In accordance with #199 I would suggest to use the non polling behavior and remove the other one. 

Perhaps someone could point out a use case where one or the other is preferable?

firesurfer, 2024-12-02:
Only place in ros-controls the `NON_POLLING` define is set is in the tests of this repository:

https://github.com/search?q=org%3Aros-controls%20NON_POLLING&type=code

If I search more general I can only find it in this repo and forks of it 
https://github.com/search?q=+NON_POLLING+AND+realtime_tools&type=code

bmagyar, 2024-12-18:
Sorry for being late on this, @christophfroehlich has brought this to the Working Group meeting. 

I also agree that we should strive for using non-polling everywhere, especially now that it can be implemented in an elegant and transparent way.

closed 2025-02-19 by destogl