ros-controls/realtime_tools #241: Is RealtimeClock() still needed with ROS 2?

christophfroehlich, 2024-12-18:
We don't use it anywhere.

Reading the discussion from #23, are those now realtime safe?

- `std::chrono::steady_clock::now()`
- `rclcpp::Clock::SharedPtr rclcpp::node_interfaces::NodeClockInterface::get_clock()`-> now()

christophfroehlich, 2024-12-18:
https://discourse.ros.org/t/ros-2-time-vs-std-chrono/6293

christophfroehlich, 2024-12-18:
copilot says

> The function `std::chrono::steady_clock::now()` is generally considered to be real-time safe. The `steady_clock` is designed to be monotonic, meaning it won't be affected by changes in the system clock (e.g., daylight saving time adjustments or manual changes to the system time). This makes it suitable for measuring intervals in real-time systems.
> 
> However, the real-time safety of `std::chrono::steady_clock::now()` can depend on the specific implementation and the underlying hardware. In most standard library implementations, it is designed to be efficient and should not block or cause significant delays, making it appropriate for real-time applications.
>

bmagyar, 2024-12-19:
Less code, healthier repo. Let's remove it

saikishor, 2024-12-19:
I think by default the rclcpp STEADY clock might be real-time safe. We have to check it properly. I mean calling stead clock and rclcpp::Clock to see if we get same information. 

However, we cannot use steady clock everywhere in the controllers due to the ROS clock, as it is mostly needed for simulation. Instead with stead clock you only have wall time

christophfroehlich, 2024-12-19:
> I think by default the rclcpp STEADY clock might be real-time safe.
You mean that [ROS_TIME](https://design.ros2.org/articles/clock_and_time.html) is not?

> We have to check it properly. I mean calling stead clock and rclcpp::Clock to see if we get same information.
But this is not the point of this issue, because the RealtimeClock is only a wrapper around a rclcpp::Clock, the user has to decide which clock_type to pass into it.

> However, we cannot use steady clock everywhere in the controllers due to the ROS clock, as it is mostly needed for simulation. Instead with stead clock you only have wall time
I agree.

saikishor, 2024-12-19:
> You mean that [SystemTime](https://design.ros2.org/articles/clock_and_time.html) is not?

SystemTime is not monotonic so it is better not used in the RT applications.

If no other repo is using it, we can remove it

christophfroehlich, 2024-12-20:
I've searched on github for it and haven't found any occurrences except of forks of this repo. I'll deprecate it for one release cycle

saikishor, 2024-12-20:
> I've searched on github for it and haven't found any occurrences except of forks of this repo. I'll deprecate it for one release cycle

Make sense. Let's do that!

PR #261 (2025-01-30): Remove RealtimeClock
RealtimeClock got deprecated on Jazzy and Rolling with their sync within the last days.

Fixes #241 and closes #23
closed 2025-01-30 by christophfroehlich