ros-controls/realtime_tools #257: PREEMPT_RT detection on kernel 6.12

alexswerner, 2025-01-15:
Hi,

to see how things go when using a mainline kernel which has most of the PREEMPT_RT patch merged I'm trying kernel version 6.12. However, the realtime_tools does not detect that it is a kernel with PREEMPT_RT:

```
[ros2_control_node-1] [INFO] [1736951683.409251251] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1736951683.409437310] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
```

The detection happens in https://github.com/ros-controls/realtime_tools/blob/5b85370c75371a5748b790101802910e53bb12f0/src/realtime_helpers.cpp#L46 through checking the existence of a file `/sys/kernel/realtime`. This file does not exists in a 6.12 kernel with PREEMPT_RT activated. Other have noticed this as well: https://forum.linuxcnc.org/38-general-linuxcnc-questions/54542-real-time-kerel-not-detected-on-patched-6-12?start=40 . In that thread people point out that this patch creates the file:
https://git.kernel.org/pub/scm/linux/kernel/git/rt/linux-rt-devel.git/commit/?h=linux-6.12.y-rt-rebase&id=3e7545de923c4e8184c2be0b95341fdc7ef1c193 . This has not been merged for now (v6.13-rc7).

saikishor, 2025-01-19:
Hello @alexswerner!

Thanks for raising this issue. I've tried a different approach in PR: #260. Do you think this will solve the issue?. If the changes look good to you, do you mind testing it and approving the PR.

Thank you!

alexswerner, 2025-01-20:
Hi @saikishor ,

I think your approach in general makes sense. I've ask the rt-users mailing list what their perspective on this is but didn't get a response (https://lore.kernel.org/linux-rt-users/70b66c69bdd3d4cf8dd7a7f17051b074b9fbcf48.camel@alexanderwerner.net/T/#u) . Maybe the lkml is actually to discuss this question.

saikishor, 2025-01-20:
@alexswerner Thanks for the response. I believe the above one should detect the RT kernel without issues. I would say this is a better approach than the earlier one, as the uname always has this information.

Thank you.

closed 2025-01-29 by christophfroehlich