ros-controls/realtime_tools #382: Please add realtime_clock back

borongyuan, 2025-08-08:
Hi, @christophfroehlich, @bmagyar, @saikishor, I just noticed that you removed realtime_clock because you thought no one was using it. But we actually use it in our VSLAM system all the time. This is the amazing performance our system has demonstrated so far.
[Best Visual SLAM Solution, Localize Disney's TRON Lightcycle Roller Coaster in the Dark in Real Time](https://www.youtube.com/watch?v=tIaFOCzrsYE)
To realize such a robust SLAM system, it is very important to correctly handle the time synchronization of sensors. realtime_clock plays an important role in our implementation. It maintains a monotonic clock while also trying to keep it synchronized with the host time (UTC time). Specifically, the OAK camera we use has an internal monotonic clock for time stamping the sensor data. When we receive the data, we need to change its timestamp to UTC time so that downstream tasks can use the data, while also maintaining the monotonicity of the timestamp. This is exactly what realtime_clock provides.
So if it's convenient for you guys, please add it back. The SLAM system is essentially a state observer, which does not go beyond the scope of the control system, so realtime_clock is suitable to be retained in this project.

saikishor, 2025-08-08:
Hello @borongyuan!

Does the RCL_STEADY_TIME work for you?, you can choose this at the time of creating your clock, this is a monotonic clock, but it is not slew-free.

borongyuan, 2025-08-12:
Thanks for the clarification. I hadn't noticed issue #23 before. This is probably because the data frequency used by our VIO is not high enough, the IMU data is only 200 Hz, so we did not encounter this problem. I took another closer look at the implementation of realtime_clock. Indeed, if we want to add it back, it would be best to provide a more reasonable implementation. I'll re-examine its original design goals and consider how to deal with this.